bummbummgarage/bummbummgarage.github.io #7: [ENVF Rev3] Big Caps are a bit wider than the panel and PCBs

steffenlohaus, 2022-01-17:


pansapiens, 2022-02-06:
I noticed this too. I did a very quick Mouser search and it seems that lowering the value to 47uF wouldn't easily provide a smaller physical size - probably only lowering the rated voltage from 35V would easily get a physically smaller electrolytic, which isn't really desirable.

I think the easiest solution would be to flip one capacitor to the other side of the board so that it's on the same side as the power connector - when bent over it probably wouldn't be any higher than the box header.

Another possibility - optional pads for those that might want to mount SMD electrolytics instead ? These could be left unpopulated if people preferred the THT component.

steffenlohaus, 2022-02-14:
I found the solution by reducing them to 10uF (and the smaller values to 100nF). Music Thing Modular has this approach.

closed 2022-02-14 by steffenlohaus

pansapiens, 2022-02-15:
This is a good solution, 100uF is probably overkill anyway. NLC also tends to use 10uF in most designs, seems to be enough.

steffenlohaus, 2022-02-15:
@pansapiens Good to hear, thnx!